im-pro-at/cncgcodecontroller #26: Auto Leveling stops 1mm above surface without doing anything

McNugget6750, 2016-07-27:
I'm using the latest version of the CNC gCode controller and a variant of Marlin that was released for the Malyan M150 - which I adapted for my M180.
No major changes, milling works perfectly fine.

When trying to use auto leveling the controller sends the command for approaching the first point
[image]

As you can see... it stops after that without anything to do. Marlin simply drives to the last G1 command and stops 1mm above the surface. It reports "ok" and does nothing else.

This seems odd and should not happen. Any ideas?

pknoe3lh, 2016-07-27:
Hey McNugget6750,

The Software is waiting for the endstopwaring from Marlin. 
When Marlin Hits the endstop it Reports it and also Reports the Position where it happend. 
The Software is using that to calculate the next move. So if there is no endstopwaring it will not continue. 

There are a lot of Settings to control the movements during the autoleveling. You have to change it that marling will hit the endstop Minimum during the g1 command. The Software is not using any speceal command for autoleveling because the original Marlin Software did not support it.

Happy milling 
Greetings 
Patrick

closed 2016-07-27 by pknoe3lh

McNugget6750, 2016-07-27:
The last command Marlin receives is G1 with a Z height of 1mm above(!) the surface. It is impossible for Marlin to report any endstop hit at 1mm above the last known Z=0mm.
My G1 commands are allowed to go 5mm below the Z=0mm, therefore it should work in theory.

Here is what happens: The head moves to Z=10mm. After that, it move to 5.05mm in X and Y. Then it slowly lowers the endmill to Z=1mm and then it stops giving commands. I don't see the command to go below the surface to actually "find" the Z height.

Am I misunderstanding something? Maybe I make a mistake somewhere... but I can't see it.

pknoe3lh, 2016-08-04:
Can you post your Autoleveling settings?

McNugget6750, 2019-03-03:
I found the issue. Only took me 2 years. You have to set the probing depth to POSITIVE values even though the default settings are set to NEGATIVE 1 (-1.000mm). Setting it to positive 2mm works!