im-pro-at/cncgcodecontroller #27: Auto leveling Issue

Iakouben, 2016-08-25:
Hi pknoe3lh,

  I started playing with your software last night, and I tried to do the auto leveling procedure. After playing with the settings, I stumbled upon the same result as user McNugget6750 has.
My machine completes the G28 sequence, moves to [10, 10] (my defined start point for the auto leveling) and sends the command G1 Z-5, after which the procedure stops. I have tried both marlin 1.0.2 and 1.1.0 RC7, and I'm getting the same issue for both firmware versions. After aborting the auto leveling procedure, I used M119 to check if the Z probe is triggered and it is.

As far as I am aware, marlin doesn't echo the Z location, or a Z trigger event on completion of the G1 command, it will just complete the command and wait for further input. G30 is used to probe for the Z location, and it will echo at what point the Z axis was hit at.

I'm not sure if I'm doing something wrong though.

Best regards,
Nikolai Iakoubenia.

pknoe3lh, 2016-08-31:
Hello lakouben,

I programmed the auto leveling long time ago ...
I use the following Firmware:
https://github.com/pknoe3lh/Marlin
My last commit was Oct 31, 2015 so its far behind....

My Marlin version echo a end stop hit event ....  
I cannot upgrade because i use a lot of modified stuff ...

Can you send me a correct sequence what the application have to send to get the hit report. 
And what it will get when there was no hit?
Then i can modify the code ;-)

I now playing around with grbl so there will hopfully be an update soon :-)

greetings 
Patrick

By the way you also always welcome to contribute code changes!

skulbl4k4, 2016-11-08:
Hi, i have the same problème, :( its a good contrôler. you are fixe the problème ?

see you soon.

pknoe3lh, 2016-12-28:
I used Auto leveling again and had not problems!
Please send me the setup and settings and i can try to understand the problem.

greetings
Patrick

skulbl4k4, 2017-01-05:
Hi,you can send the firmware (marlin and cnc gcode controlleur) for the lasted version of your CNC ?
thanks ! De Blaireville Gaël Chemin du Ruz-d'Agiez 16B  1350 Orbe  +41 79 391 74 01 theboum69@yahoo.fr

theinlinaung2010, 2017-03-20:
I also have the auto levelling issue. I can home the Z axis but the Z carriage still descends despite the trigger of the probe. I am using Marlin.

pknoe3lh, 2017-03-22:
Ok I dont have a Marlin Setup.

So can you tell me what commands are needed and what response is given.

greetings
Patrick

tomtenberge, 2019-10-03:
such a pitty noone followed up on this :(
the command for marlin is "G30"

Picture of the communications : (https://i.gyazo.com/7e87c0b7b6f0b0dc10f39a078f7617fa.png)

hopefully this helps
 ps, this is with a offset of 0 on the probe

MattRSON, 2020-05-04:
Is there any update with this? I am having the same problem with it probing one point then sitting there till it times out. I have tried G30 and I think that is the way to go but don't know how to implement it so that it uses the echo and moves to the next point.

McNugget6750, 2022-01-06:
I'm forcing my way through this as your software has the only actually working auto-level feature for pcb milling I have found so far.
I'm using GRBL-MEGA 1.1 latest.

Sequence:

1. Go manually close to the surface using the jog controlls
2. G92Z0
3. Lift 5mm
4. hit auto level start.
[image]
5. The head now moves down slowly until it hits the surface after which the following ALERT appears which also disconnects from GRBL (a feature that I'd like to remove. Disconnecting at any alert or error is not a good idea). ""Timeout: No end stop hit!""

I tracked down the error to JPanelAutoLevel.java but I can't work out how the system processes the responses from GRBL to determine that there was a contact or not. Could you please help me find the code section I should be looking?

One note: In GRBL 1.1 the probe pin reporting is different from 0.9 and earlier. It directly reports that a probe pin was hit, not a Z-axis endstop. I assume the issues could be there.

GRBL reports the hit correctly but the hit is not understood:
[image]

Any ideas?

im-pro-at, 2022-01-06:
Hallo,

I upgraded to GRPL 1.1 and the only thing that was not working out of the box was to get the position update!
I fixed that right now.

You can also see in the log "INFO: Received Z hit at: -22.1620"
So the system is reporting it correctly!

Then its also correctly Asking for the Position!
 
[image]

But it does not report it and this should be fixed now ;-)

greetings
Patrick

im-pro-at, 2022-01-06:
Oh you also need to set the following settings in GRBL to report the WPos:
$10=2

McNugget6750, 2022-01-06:
Hey Patrick,

Excellent! I pulled your last changes and it works! Thank you very much for checking this out for quickly!

im-pro-at, 2022-01-06:
You are welcome ;-)

closed 2022-01-06 by im-pro-at