im-pro-at/cncgcodecontroller #28: Different travelspeed settings for different axes

J4363R, 2017-01-06:
Hi pknoe3lh,

would it possible to add different travel speed settings for the axes or to allow the M-Code (e.g M203) to change it with the G-Code?

Best regards,
J4363R

pknoe3lh, 2017-01-06:
Hey J4363R,

what is the use case?
GRBL & Marlin support setting the maximal feedrates.

greetings
Patrick

J4363R, 2017-01-07:
Hi pknoe3lh,

In the CNC-GCode-Controller there is a setting where you can change the fast move feedrate, but you could only set the value for all axes with the same value and the problem is that the x- & y-axes of my CNC Router could move a lot faster than the z-axis. Oh, and I'am using Marlin, if you need this information. It would be nice if you could add two other buttons so it is possible to set different travelspeeds for the different axes. Or you could add support for the M-Code which sets the fast move speedrate, which would also work for my case.

Best regards,
J4363R

pknoe3lh, 2017-01-07:
OH I SEE 
1. It is a good idear to use the M203 therefore Marlin will caluclate the maximum possible feadrate for you. 
2. M203 is not a command used in normal operation ... So its not suppprted. Marlin can save the values in eeprom and so you just have to set them once 

So to set M203 just use the communication window. Here you can send what you want ;-)

Greetings 
Patrick

closed 2017-01-07 by pknoe3lh