im-pro-at/cncgcodecontroller #31: Auto Leveling - All points read the same height.

ShaneOberloier, 2017-08-01:
I'm using Cnc Gcode Controller with a Marlin based system. The auto-leveling is returning the same value for the same point. I can see that the points are being probed properly in the communication log (IE Info Received Z-hit at: XXX). The points displayed on the grid, however, are all the same value.

ShaneOberloier, 2017-08-01:
I am still looking into this myself. It may be a Marlin issue. It appears that when G1 moves Z downwards, if there is a collision, the position in Marlin (using an M114 command) reads as if it made it to the destination, and does not adjust to the Z hit position. It looks like in the code it is manually changing the position (G92), but it doesn't appear to work.

pknoe3lh, 2017-08-01:
I have no working Marlin setup at the moment :-( 
I cannot test it.
Marlin changed a lot in the last year 
So it can be a problem.
When you tell me what to change I can code the change for you ;-)