im-pro-at/cncgcodecontroller #44: Program closes after clicking "Connect"

dallonjp, 2021-05-01:
I am running the program on Windows 10 using the Java Runtime Environment 8. I have a RAMPS 1.4 shield and an arduino mega with Marlin firmware. Whether or not it is connected to the serial COM port, as soon as I select any of the viable baudrates and click "connect", the program immediately closes.

im-pro-at, 2021-05-01:
Hello,

what processor does your PC have?

greetings
Patrick

dallonjp, 2021-05-01:
Intel i5-3437U

im-pro-at, 2021-05-02:
This should work!

Java does not natively support Comports.
Therefore two libs are used: nrjavaserial-3.9.3.jar and jssc-2.8.0-javadoc.jar
This libs load a native binary and allow access to  Comport.

I did not update them in years ... So i was thinking your are maybe running ARM :-)

The solution is to just update them. 
On my PC WIN10 / Intel cpu it is running without a problem.

greetings
Patrick